kbr-net/sdrive-max #9: HX8347G LCD driver Support

iCityWork, 2018-09-21:
Any chance this can be added?
Thanks,
Gavin

kbr-net, 2018-09-25:
There is already a firmware for it included, have you tried it?

iCityWork, 2018-09-25:
I tried the normal burn/install and the display is just blank.  Doesn't seem to be supported, is there something else I need to do to make it work?
Thanks,
Gavin

iCityWork, 2018-09-25:
P.S. I am a novice when it comes to Arduino...

kbr-net, 2018-09-25:
10180  2018-09-04 23:33   atmega328-hx8347g/eeprom_writer.hex
    85231  2018-09-04 23:33   atmega328-hx8347g/SDrive.hex

You tried this files?
Maybe there is something broken, because i don't have such a display to always test the changes...

iCityWork, 2018-09-25:
I will try them tonight and let you know. 
Thanks!
Gavin

iCityWork, 2018-09-25:
Ok, I tried it, both version 0.9 and 0.8 and the display comes on, but everything is in mirror image, inverted, and when I copy SDrive.hex it starts the screen calibration, but it does not respond to touch so I can't calibrate it.
Any other ideas? 
Thank you again,
Gavin

kbr-net, 2018-09-28:
Maybe there are more variants of this display. Do you have a datasheet for it, or a pin description?

chuckcaley, 2018-12-16:
I have the same problem. it the id is 9595 its a 8347i

kbr-net, 2018-12-17:
Yes, this is slightly different, i will see, if i can make one more firmware version for it.
But first we need to know, which lines the touchpad uses. Do you have the arduino IDE installed and can try [this](https://github.com/prenticedavid/MCUFRIEND_kbv/blob/master/examples/diagnose_Touchpins/diagnose_Touchpins.ino) sketch?

kbr-net, 2018-12-17:
Let's have a try: https://www.file-upload.net/download-13436151/SDrive.hex.html

chuckcaley, 2018-12-18:
Its looking good. V1.0b, Touch screen working :) and all the writing and menu's are the right way :)
Will give it a good run through later but sofar everything is working great. 

Thanks :)
and Thanks again for the fast replys and new screen update.

kbr-net, 2018-12-18:
Nice, i used the same orientation settings as for ILI9341. Amazing, how much variants of displays are in the wild...
At last, could you check, what is printed as TFT-ID on startup?

kbr-net, 2018-12-18:
Thanks, then it seems all working as expected. When everything would be so easy :)

iCityWork, 2018-12-19:
> Let's have a try: https://www.file-upload.net/download-13436151/SDrive.hex.html

What is this junk link?? Trying to install spyware or something? Can you post the new file to try on github please?

iCityWork, 2018-12-19:
> Let's have a try: https://www.file-upload.net/download-13436151/SDrive.hex.html

Why is it reporting that this file is spyware?? Just post the new hex file on github please!

kbr-net, 2018-12-19:
Sorry, this was a development test version only, and i found no way do add attachements here.
Here  is the pre-release: https://github.com/kbr-net/sdrive-max/releases/tag/v1.0b2

iCityWork, 2018-12-19:
> y, this was a development test version only, and i found no way do add attachements here

I assumed, since I started this, you would like me to test, and it was the sketchy site you had it uploaded to that was included spyware.
Thanks,
Gavin.

iCityWork, 2018-12-23:
FYI, this does, in fact, solve the issue I was having with that particular screen, thank you very much.

closed 2018-12-24 by kbr-net