kbr-net/sdrive-max #10: Add pause between cross-hair display during screen calibration

e474, 2018-09-26:
Screen calibration can be difficult as screen can be sensitive to double taps. Incorrect calibration can be mistaken for a damaged/broken screen, which is unfortunate as screens are usually new/untested. 

Please add a pause (1 - 2 seconds) between displaying a cross-hair (or reading the screen) after each cross-hair is read or displayed.

Also, maybe display a note on the screen during calibration that calibration can be repeated by pressing screen during power up, and add a note to documentation/readme?

AtariAndre, 2018-09-28:
Her is my proposed solution (three added lines in **BOLD**

unsigned int action_cal () {
int px1,px2,py1,py2,diff;
const unsigned int b = 20; //offset from border
unsigned int x = b;
unsigned int y = b;
unsigned char i;
 
TFT_fill(Black);
 
for(i = 0; i < 4; i++) {
Draw_H_Line(x-10,x+10,y,White);
Draw_V_Line(x,y-10,y+10,White);
while(isTouching()); //wait for release
while(!isTouching());
p = getRawPoint();
**TFT_fill(Green);
delay(1000);
TFT_fill(Black);**
switch (i) {
case 0:
px1 = p.x;
py1 = p.y;
x = tft.width-b;
break;
;;
case 1:
px2 = p.x;
py1 = (py1 + p.y) / 2; //middle
y = tft.heigth-b;
break;
;;
case 2:
px2 = (px2 + p.x) / 2; //middle
py2 = p.y;
x = b;
break;
;;
case 3:
px1 = (px1 + p.x) / 2; //middle
py2 = (py2 + p.y) / 2; //middle
}
//print_I(10,50,1,White,Black,p.x);
//print_I(40,50,1,White,Black,p.y);
}

kbr-net, 2018-10-03:
Thank you, this makes sense, but i solved it my way: https://github.com/kbr-net/sdrive-max/commit/f163b736e0c380b69c3271ef882ac04b72769175

There is already a hint in the README:

> Display usage:
> 
> - on first start follow touching the calibration crosses. You can redo this
>   at any time by hold touching during startup. This have to be also done,
>   if you have changed the display rotation.

closed 2018-10-22 by kbr-net