kbr-net/sdrive-max #11: Change to remove need for diode on TxD

darylrichards, 2018-10-20:
diff --git a/SDrive.c b/SDrive.c
index 0590906..7b28795 100644
--- a/SDrive.c
+++ b/SDrive.c
@@ -374,6 +374,9 @@ int main(void)
        CMD_PORTREG |= 1 << CMD_PIN;    // with pullup
        CMD_DDR &= ~(1 << CMD_PIN);     // to input
 
+       // enable port D pin pull-ups
+       PORTD = 0xFF;
+
        //interrupts
        PCICR = (1<<PCIE1);
        PCMSK1 = (1<<PCINT13);          // for CMD_PIN
@@ -746,6 +749,7 @@ ISR(PCINT1_vect)
 
        cmd_buf.cmd = 0;                //clear cmd to allow read from atari
        process_command();
+       UCSR0B = (1<<RXEN0);            // turn off transmitter
        LED_GREEN_OFF(virtual_drive_number);  // LED OFF
 
        vp = FileInfo.vDisk;            //save actual vDisk pointer
diff --git a/usart.c b/usart.c
index 7d2927d..d33895c 100644
--- a/usart.c
+++ b/usart.c
@@ -65,7 +65,7 @@ void USART_Init ( u16 value ) {
        //UCSRA = (1<<UDRE);
 

        /* Enable Receiver and Transmitter */
-       UCSRB = (1<<RXEN)|(1<<TXEN);
+       UCSRB = (1<<RXEN); //|(1<<TXEN);
        /* Set frame format: 8data, 1stop bit */
        UCSRC = (1<<URSEL)|(1<<UCSZ1)|(1<<UCSZ0);
        {
@@ -75,6 +75,8 @@ void USART_Init ( u16 value ) {
 }
 
 void USART_Transmit_Byte( unsigned char data ) {
+       /* enable transmitter now */
+       UCSRB = (1<<RXEN)|(1<<TXEN);
        /* Wait for empty transmit buffer */
        while ( !( UCSRA & (1<<UDRE)) ) ;

kbr-net, 2018-10-22:
I'm not sure, if this makes sense. I want to be compatible to the original sdrive from Bob & Raster, and there was also the diode present. Otherways we are near the end of flash, so i won't waste any bytes not really needed.

darylrichards, 2018-10-22:
This only adds 6 bytes to the code total. Having the diode in place does no harm with this change, and I feel that since anyone new has to add the diode and this removes that requirement, it's only a net benefit.

Also, I've already remove hundreds of bytes of unneeded code from the tree that I'm working on..

kbr-net, 2018-10-27:
The diode is only the half rate, the main issue is the 1K resistor in the line, which is between the USB2Serial converter. So i think it is not really wholesome to activate a second pull-up.

darylrichards, 2018-10-27:
The optimization comes from the fact that avr-gcc isn't always great at optimizing things. Restructuring code helps it a lot... As for the pull-up, it hasn't made a difference for me either way, but since I also moved the Command signal to INT0 (PORTD bit 2), it was easier for me to turn the pull ups all on. And the reason for moving Command is again, to save a couple bytes and a slight savings in interrupt latency..

kbr-net, 2018-10-27:
But this Pin is also used for display data, you should always turn of interrupts on display activity at least.

kbr-net, 2019-04-30:
I close this issue now, because this is not the solution, and there are an other/better hardware solution based on 7407 now.

closed 2019-04-30 by kbr-net