kbr-net/sdrive-max #12: Physical drive present

e474, 2018-11-29:
This state should be added to the display for each of the SDrive drives D1 - D4 (color coding/icon would be nice).

On power up, any communication on the SIO bus that is acknowledged by another device should result in marking the corresponding SDrive Max device as "Physical Drive Present" (or similar message). Ideally this would prevent mounting images on the virtual device, or sending acknowledgement responses.

Not sure if additional code to poll devices would be practical if power is not detected from Atari computer.

kbr-net, 2018-12-17:
Don't understand, what the problem is, configure the bootD1 option and leave all drives empty, then it powers up with no drive active and no response of anything.

kbr-net, 2019-04-30:
I guess, this was an understanding mistake only.

closed 2019-04-30 by kbr-net