kbr-net/sdrive-max #17: Creating New disks doesn't respect drive setting.

ChoccyHobNob, 2019-06-28:
Creating a new disk should respect the disk setting in CFG (810 if not selected, 1050 if selected) but it doesn't. 

Creating a new disk creates a notional disk that only gets physically created when formatted. If you format with DOS2, you get 90K, if you format with DOS2.5 you get 130K regardless of what drive you have selected.

LameLefty, 2019-06-29:
As well, the SDrive-MAX firmware should recognize and respond to the standard SpartaDOS X FORMAT commands for creating a NEW disk image. Currently this fails  However, if I first boot from Atari DOS, then create a NEW image and format as either a single or enhanced density disk form Atari DOS, I can then boot SpartaDOS X, and reformat that same container ATR file with the SpartaDOS X FORMAT command, but I cannot reformat that ATR to double-density.

kbr-net, 2019-07-11:
The drive setting in CFG is only relevant to timing parameters for ATX files.
The NEW file option detects the format on the format command, which is the atari sending.
DD is not supported yet.

closed 2019-08-25 by kbr-net

kbr-net, 2019-08-27:
This may be an other issue, than the first poster entered!