kbr-net/sdrive-max #18: Screen Colour inverted? atmega328-ili9340

mamejay, 2019-07-31:
I am using the atmega328-ili9340 which is the only one that works with my display.  The background is white.
When i tested with the other builds the background is black.  Is there anyway to invert the colours on this build?
Thanks

kbr-net, 2019-08-08:
Have you testet the latest beta release 1.2b1?

mamejay, 2019-08-08:
Yes but it would not work.
My screen is a atmega328-ili9340 and the 12b1 version says its for atmega328-hx8347d
The other versions are the correct colour but the touch screen does not work

mamejay, 2019-08-08:
atmega328-ili9329 works with correct colours but all text is mirrored/back to front.

kbr-net, 2019-08-08:
Why are you sure, it is an ili9340?
ili9341looks correct, except touchscreen?

mamejay, 2019-08-08:
I went through each folder and tested installing.  The only version that gave me a readable screen and working touch was the ili9340.
ili9341 gives me correct colour but I cannot calibrate the screen and the atmega328-ili932 lets me calibrate but everything is mirrored.

kbr-net, 2019-08-08:
What ID is shown shortly after power up?

mamejay, 2019-08-08:
Its flashes up so fast but looks like c3c3

kbr-net, 2019-08-08:
Weird. If i make a Version with ili9341 screen and ili9340 touch, this would be correct?

mamejay, 2019-08-08:
I suspect that should work.  I just wrote the ili9341 code back on and can read the ID better.  d3d3.

kbr-net, 2019-08-08:
mcufriend_kbv says:

>     if (ret == 0x00D3 || ret == 0xD3D3)
> return ret; //16-bit write-only bus

Are you sure it is a 8bit?

mamejay, 2019-08-08:
I am not sure.  Was just a screen I purchased from Banggood.  d3d3 is what shows up on bootup

kbr-net, 2019-08-08:
Try this:
[sdrive-max-v12b2.zip](https://github.com/kbr-net/sdrive-max/files/3481413/sdrive-max-v12b2.zip)

mamejay, 2019-08-08:
> 
> 
> Try this:
> [sdrive-max-v12b2.zip](https://github.com/kbr-net/sdrive-max/files/3481413/sdrive-max-v12b2.zip)

Perfect!  That works as expected now.  Thank you very much

kbr-net, 2019-08-08:
Ok, i am thinking about the final integration, maybe i will add an autodetection for the touchscreen, then you could use the ili9341 in future...

mamejay, 2019-08-08:
If you need me to test let me know

kbr-net, 2019-08-08:
You are welcome!

kbr-net, 2020-01-29:
Please try https://github.com/kbr-net/sdrive-max/releases/download/v12c2/sdrive-max-v12c2.zip

TzOk83, 2020-01-30:
Touchscreen still doesn't work.

kbr-net, 2020-01-31:
> Please try https://github.com/kbr-net/sdrive-max/releases/download/v12c2/sdrive-max-v12c2.zip

this version was buggy, please wait for v12c3, and then try this...

mamejay, 2020-03-27:
Think this one can be closed.  Everything seems to be working correctly

closed 2020-03-27 by mamejay