kbr-net/sdrive-max #19: No touch on a 9340

mamejay, 2019-08-14:
Got a couple of bonus smaller 2" screens with an order
I have tested with the different firmware but cannot get the touch to work.
The ID that appears on boot is 9340
If I use the atmega328-ili9340 the colour is correct but the picture is inverted and touch does not work.  It lets me calibrate but does not register anything after that.
Anyway to test to work out what touch is required?

kbr-net, 2019-08-25:
Have you tried the special pre-release V1.2b?

Install the arduino IDE and try this sketch: https://github.com/prenticedavid/MCUFRIEND_kbv/blob/master/examples/diagnose_Touchpins/diagnose_Touchpins.ino

kbr-net, 2019-08-25:
Or try the firmware in this issue below: https://github.com/kbr-net/sdrive-max/issues/18

mamejay, 2019-08-26:
Thanks for that.  I have tried all the firmware that is available.  I have tried the diagnose_touchpins sketch but I just get a blank screen.  Am I supposed to connect this to a serial connection to read the results?

kbr-net, 2019-08-26:
Yes, this only examines the touchscreen lines, there is no display output. Just open the serial monitor on the arduino IDE.

kbr-net, 2020-01-29:
Done in issue #18

closed 2020-01-29 by kbr-net