kbr-net/sdrive-max #20: LED

iCityWork, 2019-08-14:
Are there any ports on the Arduino that I could hook LED's to in order to have a power LED and drive activity LED? Working on an 810 enclosure and would love to have the two LEDs on the front functional.
Thanks,
Gavin

kbr-net, 2019-08-25:
Would be difficult, because all ports are in use.

iCityWork, 2019-08-25:
There is no way to piggyback on one of them to drive the LED?

kbr-net, 2019-08-26:
The only way i see, the PLCC-Version of atmega328 has 2 more ports, which are not assigned to pins, they could be used.

kbr-net, 2019-08-26:
An other idea is to use the SD cards chip select line(Pin 10) to drive a led with, but this will be very short flashes.

Mi1ezDR, 2019-12-10:
I'm still a circuit noob but couldn't you could use a capacitor to help keep the LED's lit for a little longer (milliseconds) until the capacitor has emptied out so that you have a visual chance to see it's been working?

kbr-net, 2019-12-21:
Just try it!

TzOk83, 2019-12-30:
Actually the USER_LED, which is hooked to the PIN 13 (PB5), is nicely blinking when there is a drive activity. It is the SD-card's CLK line (SD_SCK).

closed 2020-01-17 by iCityWork