kbr-net/sdrive-max #21: ili9338 touchscreen (Mcufriends)

crenedecotret, 2019-08-20:
Hello, I ordered a screen from Amazon. The pictures indicated a model that is supported but what I received is an ili9338.

I tried flashing all the ili files with no luck.. the screen remains white all the time.

I'll be returning the screen to amazon soon but I figured I would try logging an issue first to see if a fix is possible.    thanks!

kbr-net, 2019-08-25:
Without a datasheet, and a pin assignment, this would be difficult. Can you post a picture from the backside of the display?

crenedecotret, 2019-08-25:
[image]

crenedecotret, 2019-08-25:
Here you go... I fully expect to send it back soon. I have a couple of screens that specifically mention the chipset on the way from aliexpress (came out cheaper than this one from amazon). My reasoning opening the ticket was that maybe it could help someone out and I can test while I still have the screen here

closed 2019-08-25 by crenedecotret

kbr-net, 2019-08-26:
Thanks, the pin assignment looks correkt, but if there is no output either, this seems to use a completely other protocol, which would be too much work to implement.
An other way was to try get it working with the standard arduino libraries from mcufriends, and then find out, which commands it needs...

crenedecotret, 2019-08-29:
thanks for looking into it!