kbr-net/sdrive-max #22: How to add support to a new display?

jfsantos, 2019-09-01:
I have a display with a HX8347D and it doesn't work with the support you added for the G and I variants of that controller. I have another library that works with that display, but the API is very different from yours. How should I go about adding support to this controller?

kbr-net, 2019-09-01:
You can do it by yourself and enter a pull request, or show me the library, and i will see to add this.

jfsantos, 2019-09-02:
I don't know how to best integrate it to your codebase as I haven't worked
with TFT displays before and I don't know what they have in common, but I'd
be happy to help test it. This is the manufacturer's code:
https://www.waveshare.com/wiki/File:2.8inch_TFT_Touch_Shield_code.7z.

kbr-net, 2019-09-02:
Thank you, but i am sorry, this display uses a completely other controlling via SPI. I only want to support 8bit parallel display interface and analog touchscreen. Any other variant does not fit in the current concept, this would be too much work and SPI would be very slow, so we will get timing problems on SIO port also and so on...

Sorry again...

closed 2019-09-02 by kbr-net