kbr-net/sdrive-max #24: Supported display types

pavelru, 2019-10-18:
Hello, I bought a display on the e-shop website https://arduino-shop.cz/arduino/1568-shield-lcd-tft-2-4-display-for-arduino-uno.html unfortunately not supported by you . I found that the following demo works on the recommendation of the e-shop https://arduino-shop.cz/docs/products/0/141/test_example_1487762760.zip. Could you include support for this type of display? Thank you very much for your time.

pavelru, 2019-10-19:
Just add that the display correctly displays the screen after writing the firmware atmega328-ili9329 but after performing the display calibration, the touch control stops working.
After writing the firmware version of atmega328-ili9340, the screen is upside down and mirrored.
And after writing the firmware version of atmega328-ili9341eeprom_writer.hex, the screen is mirrored, and after writing sdrive.hex the display cannot be calibrated. The touch control does not work.
[image]
[image]
[image]
[image]
[image]

pavelru, 2019-10-19:
I will add one more clarification. I recorded the progress of writing sdrive.hex on my mobile phone and after downloading any version I will briefly flash TFT ID: 9340 ... Then I do not understand why my version of firmware atmega328-ili9340 does not work correctly :(

kbr-net, 2019-10-19:
Please read https://github.com/kbr-net/sdrive-max/issues/18 and try the firmware v1.2b2 there.

pavelru, 2019-10-19:
Good morning and thank you for your answer. I tried to write this version of atmega328-ili9340i firmware as well. The display is shown in the attached photos.
[image]
[image]

pavelru, 2019-10-19:
I also run pin diagnostics see below if it helps.

Making all control and bus pins INPUT_PULLUP
Typical 30k Analog pullup with corresponding pin
would read low when digital is written LOW
e.g. reads ~25 for 300R X direction
e.g. reads ~30 for 500R Y direction

Testing : (A1, D7) = 32
Testing : (A2, D6) = 25
Diagnosing as:-
YP,YM:  (A1, D7) = 32
XM,XP:  (A2, D6) = 25

kbr-net, 2019-10-19:
Weird, this are the correct pins for ILI9329 and ILI9340. Calibration works, and then it hangs?
Then the problem is not the touchscreen, i think.
What happens when the Atari is powered on?

pavelru, 2019-10-20:
Hello, I enclose photos of the display and cable connection to the connector and to Arduino. The SD card is formatted by fat or fat32, and the sdrive.atr file is in the card's root. Unfortunately, nothing happens when you turn on Atari. Just growls and BASIC starts. Calibration is counterclockwise as shown in other photos.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

kbr-net, 2019-10-20:
Try with ILI9329 firmware, this looks the best, and also try to start Atari after calibration without display attached.

kbr-net, 2019-10-20:
There is also no response on touch if SD-Card is out and SIO not plugged in?

pavelru, 2019-10-20:
I installed the sdrive-max-v11\atmega328-ili9329 firmware and performed the steps you recommended. Even then, nothing happened. So I tried to use another SD card and sdrive.atr is loaded while Option is pressed. For some reason I don't want to work with a small 128MB SD card. Unfortunately, the touch control still does not work. But when I leave sdrive for some time and the screensaver comes up, so by touching the screen wakes up ...
[image]

kbr-net, 2019-10-20:
Can you send me the first 2MB raw data of the SD-Card?
On Linux you can do this with: dd if=/dev/sdc of=sd.out bs=1M count=2; gzip sd.out
(where /dev/sdc is your card device)

pavelru, 2019-10-20:
I would send but unfortunately I do not use linux :( Is there any possibility to do this on win10?

kbr-net, 2019-10-20:
Maybe there are tools for win10, but i don't know about, sorry. Use a KNOPPIX CD?

I have also released a new beta v1.2b3 right now, please try: https://github.com/kbr-net/sdrive-max/releases/download/v1.2b3/sdrive-max-v12b3.zip

pavelru, 2019-10-20:
Unfortunately, the new firmware is mirrored. the touch control works very strangely.
[image]
[image]
[image]

Tomorrow I will try to prepare an old NB with Linux and try to send the required files

kbr-net, 2019-10-20:
I'm a little bit confused now, in you second post the ILI9329 looks correct, which firmware is now with the correct display?

pavelru, 2019-10-20:
The last firmware I installed was beta sdrive-max-v12b3.zip

kbr-net, 2019-10-20:
Sorry, my fault, please try the same file again.

pavelru, 2019-10-20:
Nothing happens. I tried the latest firmware. Displays correctly and touch control works but as if for a large display. If I touch the middle of the screen at some point, I get to the file selection. But ... I sometimes get to the READY screen at the same place.

kbr-net, 2019-10-20:
Can you post a picture with the values in the middle at the end of calibration again?
You can initiate a recalibration by holding touch while power on.

pavelru, 2019-10-21:
Good morning, I'm sending the photo.
[image]

kbr-net, 2019-10-21:
Try to clear the bootblock of the problematic SD-Card, create a new partition table, and format it new.

The values of the calibration looks good, don't know, why it's not working...
Have made a firmware with touchpoint debug X, Y output in pixels. Left upper corner should be near 0, 0, right bottum 240, 320. Hope that helps!
[sdrive-max-v12b3-debug.zip](https://github.com/kbr-net/sdrive-max/files/3751780/sdrive-max-v12b3-debug.zip)

pavelru, 2019-10-21:
Good evening, thank you very much for your help. I send a photo and the next one is photographed the problem, why it does not work as it should. If I press the joint a little with my finger, the touch control works. I will have to claim the display from the seller. Once again, thank you very much for your continued help. If you can send a contribution, please write.
[image]
[image]

kbr-net, 2019-10-21:
All right, but the firmware for ILI9329i is now the correct one?

pavelru, 2019-10-22:
Yes, touch control worked as far as possible. I will probably confirm tomorrow when I receive a new display.

pavelru, 2019-10-23:
Hello. Today I received a new display. After connecting to arduin, start up and recalibrate works at first good. I tried to upload a version of sdrive-max-v11\atmega328-ili9329, but it does not work after calibration touch control. I returned to the version of atmega328-ili9329i attaching the picture. Thank you very much again for your time.
[image]

Now, after attaching the disk image and pressing Option + Reset, it displays this message as shown. He tried 2 SD cards and it is the same for both.
[image]

kbr-net, 2019-10-23:
As far as i can see, you have connected the SIO power pin 10 to 5V, and also connected USB, this is no good idea, use only one power source!
And then try power on by holding OPTION, RESET does not always work correctly.

pavelru, 2019-10-23:
The first photo is sdrive connected to the PC and the second to the atari without additional usb power. The cable shown in the second photo charges my tablet.

kbr-net, 2019-10-23:
But sdrive.atr is loading the control program?

pavelru, 2019-10-23:
Sdrive.atr will load correctly. I select a disk image in D1 and the message appears after pressing Opt + Reset.

kbr-net, 2019-10-23:
Strange, "CMD Timeout" says, the CMD-Line is pulled low, but no command is sent to the SDM.

kbr-net, 2019-10-23:
Can you see the directory of the disk image in the control program by pressing "V"?

pavelru, 2019-10-23:
In the control program, after pressing L and I, I see a long file name and information and file. After the V label, the screen is blank, but the load indicator glows green. Press V to return to the control program.

pavelru, 2019-10-23:
This will appear when sdrive.atr is loaded.
[image]

kbr-net, 2019-10-23:
> In the control program, after pressing L and I, I see a long file name and file information. After the V label, the screen is blank, but the load indicator glows green. Press V to return to the control program.

Show me this output please, if you have selected the image, which won't boot.

pavelru, 2019-10-24:
Hello, I'm sending a link to the video where you can see how it behaved. When I am newly downloaded BiboDos7 already works properly. The error was probably caused by copying to the SD card and disconnecting it incorrectly from the system.
https://photos.app.goo.gl/g5AVPu3a7DT9qSzL9

kbr-net, 2019-10-24:
Thank you, then we can close this issue, i think.

closed 2019-10-24 by kbr-net