kbr-net/sdrive-max #25: help HX8347I

ascrnet, 2019-11-27:
Hi all,

I have an HX8347I screen which flickers a lot using the v1.0 or v1.1 version connected to the notebook since I still don't make the cable, is that flicker normal that seems to temper the main screen where the units are?

link HX8347I 
https://www.aliexpress.com/item/32905829057.html

Regards

kbr-net, 2019-12-02:
No, i think, this is not normal.

ascrnet, 2019-12-19:
connecting to atari looks worse
https://www.atariware.cl/archivos/sdrive-max/error-screen.jpg

Regards

iCityWork, 2019-12-19:
It looks like an under-voltage.  I always use an external 12VDC 1A power adaptor and I never have display issues.

ascrnet, 2019-12-19:
here I put a video to see the effect of the screen and that by clicking on READY the screen looks perfect
https://youtu.be/N5whhq6k62E

I have the impression that it is incompatible, this screen has the chips lvc245a
https://www.atariware.cl/archivos/sdrive-max/hx8347i.jpg

Regards

iCityWork, 2019-12-20:
It is power! Trust me, run it on an external power adapter and try again. I have seen that many times and that is what is causing it. The computer SIO 5VDC is such low power the Arduino and touch screen are getting too little.  Try and see if it solves that issue.

kbr-net, 2019-12-21:
I think also it is power leakage. If it was incompatible, the display would be always wrong...

ascrnet, 2019-12-21:
Thanks for all the help, it seems that this screen consumes a lot of voltage. Buy another screen ILI9341 I hope it is better and consume less voltage to use only the SIO without external sources.

Regards

closed 2019-12-21 by kbr-net