kbr-net/sdrive-max #26: XPT2046 touchscreen support

TzOk83, 2019-11-30:
There are quite popular ILI9341+XPT2046 TFT touchscreens available. Would you consider adding support for this touch controller chip?

bweiser, 2019-11-30:
The ILI9341 has support for it already.

TzOk83, 2019-11-30:
I realize too late that you only support ILI9341 over a parallel link, and not over SPI. XPT2046 is an SPI touch (digitizer) screen controller, commonly used in SPI TFT screens based on ILI9341 chips. Your code supports only direct (analog) signals from digitizer screen, but in case it doesn't support SPI mode for graphics, there is probably no point to support SPI touch sensor.

kbr-net, 2019-12-02:
Yes, this would be too difficult, and would be like a completely rewrite of the firmware.

closed 2019-12-02 by kbr-net