kbr-net/sdrive-max #28: Genneral use of ILI94XX series

bweiser, 2019-11-30:
I made a discovery that the ILI9481 - ILI9488 series TFT seems to work with the 9341 series drivers. I've yet to do a full test, but the menus and calibration work fine. The only catch is I have a 3.5 inch screen that supports 480x320 resolution, where as the default is 240x320. So I only get over 1/3 of the screen as a viewing area. (see attached)

I'm interested in having a larger touch screen to work with. It might help some of us with vision issues, or just want a bigger screen, just because we can!.  These Kuman brand are really well built to. It's not always about going for cost savings, where as it's for comfort of view or just because it looks great.. Is there an easy way to change the resolution, or offer an advanced menu on the sdmax to change the resolution, and enable a time out if it fails. One that requires accepting the image post change. 

[image]
I'll update after I get a chance to test and make sure it's 100% working normal on the 9341 loadset.

bweiser, 2019-11-30:
[3.5inch_Arduino_ILI9486_V1.0.zip](https://github.com/kbr-net/sdrive-max/files/3907486/3.5inch_Arduino_ILI9486_V1.0.zip)

kbr-net, 2019-12-02:
This would be also near a completely rewrite of the firmware, i won't do that.

closed 2019-12-02 by kbr-net