kbr-net/sdrive-max #29: Cannot get any display using any of the hex code

mamejay, 2019-12-02:
Hi there.  I grabbed more screens from the same supplier on banggood that I have bought a few times and looks like they are not the same now.
They look exactly the same  but with slightly different silkscreen on them.
I have run the mcufriend_diag_tft_support.hex and the screen works with a ID=0x6809
Any ideas of what the display is?

[image]

kbr-net, 2019-12-03:
There is a pull request waiting with firmware for ILI9325, maybe this will work with it too.

mamejay, 2019-12-03:
Thanks for that.  Github is very foreign to me unfortunately.  So someone has added support for the ILI9325?  Would there be pre-compiled HEX for it?
My display is a RM68090

mamejay, 2019-12-04:
Found the HEX of the ILI9325 and uploaded it.  Same white screen so looks like RM68090 is not 100% compatible.  Getting sick of suppliers changing their stock on each delivery

kbr-net, 2020-01-29:
Sure, this display is using the 8bit parallel mode, and not SPI?

vladkvladk, 2020-02-02:
Hi,
I have modified ILI9325 firmware so now RM68090 is supported. Unfortunately, I cannot test it with real Atari at the moment, but the display and TS both work. Try it out and let me know.
[sdrive_rm68090.zip](https://github.com/kbr-net/sdrive-max/files/4144758/sdrive_rm68090.zip)

mamejay, 2020-02-03:
Thanks for that.  The Eprom file works and gives me display but I am getting  a Not Detected on the touch screen now.  So close!!

mamejay, 2020-02-03:
Edit : My bad.  Tried a second screen and it works perfectly.  This code is working!!

kbr-net, 2020-02-04:
Nice. If you want me to add this changes, please drop me a pull request.

closed 2020-02-04 by kbr-net