kbr-net/sdrive-max #33: Missing Motor-ON line for tape emulation

TzOk83, 2019-12-19:
There is no control over the emulated tape drive motor, so it doesn't stop when it should.

kbr-net, 2019-12-21:
There is no free port for the motor signal...

TzOk83, 2019-12-21:
The Command line is not used in Tape Emulation mode, nor Motor ON line is used in Floppy Emulation mode, maybe it would be possible to use same pin (A5) for both purposes, only with a diode separation?

kbr-net, 2019-12-21:
This would be quite not so easy, because COMMAND is active low, and MOTOR is activ high!

TzOk83, 2019-12-21:
It would be a little tricky, but not impossible. Motor On is open collector, but it is Hi-Z when inactive, and Vcc when active, so it requires a pull-down. Command is driven directly by PIA, but has pull-up resistor, so it is also open-collector output. This time with "hard" ground, and "soft" Vcc. It has a 100 Ohm series resistor (making it short circuit resistant). Unfortunately ATMega 328P has only switchable pull-up resistors.

kbr-net, 2019-12-21:
But not without extra circuit. I'm not sure, if it is worth, because the tape emu is just a gimmick for me, once i had fun to make this :)
And because of low on free flash, maybe i will remove it sometimes...

TzOk83, 2019-12-22:
Actually pin D10 seems to be obsolete, there is only one SPI device (SD card) on the system, so ~SS could be hard wired to GND, and D10 freed for another (Motor On) use.

Tape emulation is a very unique feature of SDrive MAX

kbr-net, 2019-12-22:
It is necessary to toggle chip select for the SD protocol.

I think, there is no way without hardware modification, so i won't do that.

SPI for the display would be very slow, so i also won't support this.

kbr-net, 2019-12-30:
One more idea is to use a display data pin, because this pins are Z, if display is not in use. So we could connect MOTOR over a 1K resistor to one of them, and this should not hurt.
The only thing, it could make trouble with the new touchscreen autodetection i am working right now, there i have to check all this pins for resistance...

TzOk83, 2019-12-30:
For touchscreen auto detection you can "inspire" by the TouchScreen_Calibr_Native.ino from MCUFRIEND_kbv library. This one works quite ok.

kbr-net, 2022-11-04:
A new idea for this issue is to multiplex the COMMAND and MOTOR lines together, because they would never be used at the same time. Maybe...