kbr-net/sdrive-max #36: Documentation for sdrive.com??? anywhere?

jedimatt42, 2020-01-11:
I can't locate end user documentation for sdrive.com...  I thought it was built into the program.. but then I can't remember how to view it either.. 

Something in the README.md pointing to those docs would be good.

ascrnet, 2020-01-14:
> 
> 
> I can't locate end user documentation for sdrive.com... I thought it was built into the program.. but then I can't remember how to view it either..
> 
> Something in the README.md pointing to those docs would be good.

Read the following pdf contains the keys SDRIVE.COM 
https://atari8bit.net/wp-content/uploads/SDrive_en.pdf

kbr-net, 2020-01-22:
Hold down the SHIFT key while booting from D0: (SDRIVE.ATR), then press "3" to start the version with help, and you get a key description by pressing the HELP key.

kbr-net, 2020-01-29:
There are also the original docs available on Rasters website:
http://raster.infos.cz/atari/hw/sdrive/SDrive_en.pdf

closed 2020-01-29 by kbr-net

s-simpson, 2020-04-19:
https://atari8bit.net/everything-sdrive-max/

s-simpson, 2020-04-19:
https://atariwiki.org/wiki/Wiki.jsp?page=SDrive-MAX

s-simpson, 2020-04-19:
https://www.youtube.com/watch?v=-OshRXxzJTY