kbr-net/sdrive-max #37: Problem after flash.

pasmoi87, 2020-01-17:
Hi,
First flash with atmega328-ili9329i my screen is miroring, touch screen is working.
So i test atmega328-ili9340, my screen is now in good meaning, touch screen initialization working but after this touch screen not working.
I test atmega328-ili9341, the screen is in good meaning, but touch screen initialization not working.
An idea of the problem ?
Thanks.

kbr-net, 2020-01-26:
Please test the new firmware with touchscreen autodetection: https://github.com/kbr-net/sdrive-max/releases/download/v12c1/sdrive-max-v12c1.zip

pasmoi87, 2020-01-30:
Hi,
Thanks, but not working for me.
The only one that work is the atmega328-ili9329i : touch screen OK after initialization, but screen is mirroring.
With all others touch screen does'nt work.
A atmega328-ili9340i for me ??? or a atmega328-ili9329ii ??? It is a little joke if I weel understand what is the 'i', because it doesn't make sense to me, reverse the reverse.

pasmoi87, 2020-01-30:
atmega328-ili9329i or  atmega328-ili9329, screens are the same, inverted.

kbr-net, 2020-01-30:
Then test the v12c2, and show me the output screen of eeprom_writer.hex.

pasmoi87, 2020-01-30:
[image]
[image]

pasmoi87, 2020-01-30:
Can't initialyse touch screen.

kbr-net, 2020-01-31:
Hm, this looks like touchscreen is always pressed, A1 has connection to PINB0 and PINB1.
Are you sure, not have touched the screen while detection in progress?

pasmoi87, 2020-01-31:
Not touch the screen.
[image]

pasmoi87, 2020-01-31:
With ili9329i, all is fine (it works, i can push cfg, exit etc...) but screen is inverted.

kbr-net, 2020-01-31:
Thank you for debugging, i think i found the problem, please give me some time to fix this...

pasmoi87, 2020-01-31:
No problem. Thank to you, It's you who are making all the work. Take your time.

kbr-net, 2020-01-31:
Nope, that was is not..., could you please run this eeprom_writer.hex with debug, and show me the output?
[eeprom_writer.hex.gz](https://github.com/kbr-net/sdrive-max/files/4140083/eeprom_writer.hex.gz)

iCityWork, 2020-01-31:
Are you sure your TTF is working correctly? Has it been tested with anything else? I know I have had a few that have been defective and didn't work correctly.  Just a thought...

pasmoi87, 2020-01-31:
[image]

pasmoi87, 2020-01-31:
No, just buy for this.

kbr-net, 2020-01-31:
Thanks, but something in your list may be wrong, i gave you the version for ili9340, and this is inverted now, so i think you need ili9341!

pasmoi87, 2020-01-31:
[image]
For me this is that, inverted screen. touch is in the good place.

kbr-net, 2020-01-31:
This is not inverted, this is mirrored! So you need ili9329.

kbr-net, 2020-01-31:
Please try this one: 
[eeprom_writer.hex.gz](https://github.com/kbr-net/sdrive-max/files/4141324/eeprom_writer.hex.gz)

pasmoi87, 2020-02-01:
this with 1.2c2 is nice. thank. (all 1.0 and 1.1 don't work with this)
I can navigate in cfg, select or deselect item save and go out. 
Now the problem is when I insert the microsd card, touch screen doesn't work anymore.
My microsd is in FAT, with sdrive.atr in root. D0: with a punct. (without punct if no card so card is well detected)
After reset with card same thing, touch screen doesn't work.

pasmoi87, 2020-02-01:
IT'S OK, microsd in FAT32 work fine.

kbr-net, 2020-02-01:
Please try the even released v12c3, then we can close this issue.

pasmoi87, 2020-02-01:
My microsd is the problem. Thank you for your work.
You develop a SDrive for CPC but Gotek already existe and now a lot of user have this, it is not possible to do an addon like this 2,8'' screen and touch screen with a nano arduino to navigate and select a file on the usb key ?
Or this https://www.youtube.com/watch?v=qLoFTItIIGo, it was develeped for Amiga Cortex firmware, this firmware is obsolete, but it will be interresting on Flashfloppy or HxC firmware, only an ESP8266, no new of the developer since 3 years.

kbr-net, 2020-02-01:
The SDrive4CPC is a completely other concept, only the SD card is attached to the parallel port, and all other things have be done in software, so it is not possible to select images by hardware...

closed 2020-02-01 by kbr-net

pasmoi87, 2020-02-01:
Ok,  ili9340 1.2c3 work fine.