kbr-net/sdrive-max #39: Screen autodetection still doesn't work in v1.2c2.

TzOk83, 2020-01-30:
Still no-go for screen auto detection with v1.2c2...

kbr-net, 2020-01-31:
Could you post the output screen of eeprom_writer.hex please?

TzOk83, 2020-01-31:
Here you have it (this is for ILI9340):
[image]

Colors are inverted, I can't pass touchscreen calibration.

kbr-net, 2020-01-31:
Strange, this looks also like touchscreen is always pressed...
Does it work correctly with other arduino applications?

TzOk83, 2020-01-31:
Yes, it works with the example sketches from MCUFriend_kbv, as well as with SDrive MAX v1.2b with swapped X and Y axes in touchscreen (configured like for HX8347G).

kbr-net, 2020-01-31:
Could you please run this [eeprom_writer.hex.gz](https://github.com/kbr-net/sdrive-max/files/4140094/eeprom_writer.hex.gz), and show me the output?

TzOk83, 2020-01-31:
Here you have:
[image]

kbr-net, 2020-01-31:
Damn, this looks correctly...

TzOk83, 2020-01-31:
Maybe this will explain more:
[image] 
[image] 
[image] 
[image]

I believe this is not using EEPROM data, as this was executed with EEPROM content from a working v1.2b.

kbr-net, 2020-01-31:
Thanks, but this does not matter, the values without touch are necessary.

kbr-net, 2020-01-31:
Please try: 
[eeprom_writer.hex.gz](https://github.com/kbr-net/sdrive-max/files/4141327/eeprom_writer.hex.gz)

TzOk83, 2020-01-31:
Touch now works, but colors are still in negative.

kbr-net, 2020-02-01:
Hm, ok, then i have mixed something up...
I will release a new beta v12c3 soon, please test this again.

TzOk83, 2020-02-01:
v12c3 flashed as ILI3941 seems to work ok (yet on boot I have message saying TFT-ID: 3940).
Thank you.

kbr-net, 2020-02-01:
Ignore it, there are so much display variants in the wild, maybe production failures etc...

closed 2020-02-01 by kbr-net