kbr-net/sdrive-max #40: Are there any working displays with autotouch feature?

kbr-net, 2020-01-31:
Please report success, ideally with a screenshot of eeprom_writer.hex, thanks!

kbr-net, 2020-03-27:
Ok, people seems to like only report errors, so i assume they are all working :)

a8dogdark, 2020-04-05:
I with the version, atmega328-ili9341, I have not had problems, it looks good. thanks.

battlecoder, 2020-06-26:
New guy here. Was using a previous version but just updated to the latest with the auto-touch feature, and it worked wonderfully with a ILI934x LCD that I know for a fact didn't have the "standard" touchscreen pinout.
My display, however, required tweaking the LCD initialization. That's something I knew already, because I've used the same screen in the past. I do not know if the changes I had to make "break" compatibility with "regular" ILI9329 or ILI934x screens, but I don't want to derail this thread with that. I can provide further details in a separate thread if anyone is interested.

Auto-touchscreen worked, and it's definitely a great feature.

closed 2022-11-04 by kbr-net