kbr-net/sdrive-max #41: Screenless mode - button to cycle through D1 - D4

TzOk83, 2020-02-01:
It would be great, if there would be an option to use some button to cycle through drive slots. I know there is no more free GPIO pins, but there could be either alternate version (like there are for various displays) or some flag stored in EEPROM, whether screen was detected or not. In that case, is would be also nice to add 4 LEDs to indicate selected slot (like it was in the original SDrive).

kbr-net, 2020-02-01:
You know my old [sdrive-ng](http://www.kbrnet.de/projekte/sdrive-ng/index.html) project? There is all, what you want, but i don't want to port this things to SDrive-MAX, because normally we have a touchscreen, so i will let support without screen at a minimum.

closed 2020-02-01 by kbr-net

kbr-net, 2020-02-01:
You can make it by your self, and drop me a pull request, then i will merge this to the project, but i don't want to investigate time for it, because it's my spare time, and i got not payed for it, so it's my decision, what i implement, and what not.

TzOk83, 2020-02-01:
I just make suggestions, and of course you choose whether to implement them or not. I didn't know SDrive NG. Now I see it has source code available, so I'll rather go that way and make it work on Arduino Nano, without need for changing fuse bits, and with just one button.

kbr-net, 2020-02-02:
Sources are also available at github: https://github.com/kbr-net/sdrive-ng
And there is a further version [V2](https://github.com/kbr-net/sdrive-ng-v2), where i optimized the SD card connection, but then i built SDrive-MAX.