kbr-net/sdrive-max #44: Screen autodetection with sdrive-max-v12c3 atmega328-ili9329

mamejay, 2020-03-11:
I have a couple of 2.4" screens.  I can get them flashed using atmega328-ili9329 and they show touch detection but after doing the calibration I am getting no response from the screen.
Below is a screenshot after eeprom write.  Let me know if you suspect my screen is faulty or something else.

[image]

iCityWork, 2020-03-11:
I have gotten that a few times, it seems some screens just will not accept the program.  I have had it happen to me about 20 times in the last year with a little over 200 builds I felt that was an acceptable percentage of bad screens.

kbr-net, 2020-03-17:
The value of 70 looks me to high, so i also think, it's broken.
Maybe you can try an older firmware without autodetection.

mamejay, 2020-03-27:
I think you are correct.  Got more of these 2.4"screens and they are all working fine.

kbr-net, 2020-03-27:
nice :)

closed 2020-03-27 by kbr-net