kbr-net/sdrive-max #45: load cas at more than 800 baud

a8dogdark, 2020-04-05:
I was doing cas tests with the sdrive the new version in 256, and it reads well in standart speed, 600 baud, my query is the following: could the cas system be able to load an atari at more than 800 baud? cas that work at 600, 800, 1150 and 1400 baud.

https://github.com/a8dogdark/coleccion_cas/tree/master/ULTRA

In that folder you can find cas files at different speeds.

nhp 600 baudios
8nhp 800 baudios
stac 990 baudios
ultra 1150 baudios
super ultra 1400 baudios

Likewise, very grateful for the latest version, sdrive-max-v12c3, which perfectly reads 256-byte files.

kbr-net, 2020-04-07:
Included in V1.2.

closed 2020-04-07 by kbr-net

a8dogdark, 2020-04-11:
Thank you very much, I managed to make it work, although it only reads up
to 1000 baud, it is already a tremendous achievement. A hug.