kbr-net/sdrive-max #47: Using Atari DOS 2.0D  (double density version) new command creates single density sized blank disk

s-simpson, 2020-04-19:


s-simpson, 2020-04-19:
LInk for dos 2.0D

http://www.atarimania.com/utility-atari-400-800-xl-xe-dos-20d_17585.html

------------------------------------------------------------------------------------------

if you boot the Atari dos 2.0d, everything works, it loads and boots like I remember, you can save and load files onto the existing atr.  If you use the new command to make a new disk, it makes the new disk, and when you format it, the disk appears to function ok.  The file that is created, is however the wrong size.  It should be the same number of sectors as a SD disk, but 256 bytes per sector, rather than the 128 bytes per sector.  A double density blank atr file should take up double the space as a single density atr file on the sdcard.  Eventually the new blank is going to run out of actual space before it thinks its empty.  Either an option to specify which type of 'new' disk to be created is needed, or the sdrive has to detect the density of the currently running dos.  I imagine that this might affect 'virtual hard disk partitions' in sparta dos as well.  The only known current work around is to make the double density atr files on another computer and copy them in advance over to the sdcard.

s-simpson, 2020-04-19:
I think the first few sectors of a double density disk aren't 256 bytes per sector though, If memory is correct the first 3 were 128 bytes, and the rest were 256 bytes per sector, but there must be documentation somewhere to confirm this

kbr-net, 2020-04-19:
Dos 2.0d sends the regular command $21 for FORMAT SINGLE as specified, there is no percom or something else, so Sdrive-MAX is doing all right. I think, the 815 is a very special thing, and can only format DD.
Don't use this special Dos!

Spartados is working.

s-simpson, 2020-04-19:
> 
> 
> Dos 2.0d sends the regular command $21 for FORMAT SINGLE as specified, there is no percom or something else, so Sdrive-MAX is doing all right. I think, the 815 is a very special thing, and can only format DD.
> Don't use this special Dos!
> 
Ok, well it worked ok on the original percom drives, and formatted double density.  perhaps the software didn't follow later conventions as you couldn't select which density to format in that dos.  Originally i used it because nothing else existed at the time.  It was given to me as a patch to the original atari dos by percom when I bought the drive.  you had to boot the atari dos 2.0, install the patch in memory and format a new disk and then write dos files to a new disk to get the double density storage.  There must have been a utility at the time to transfer files between single and double density, but I can't remember.  It was kind of a pain.  I couldn't find anything in your code, but your explanation makes sense to me.

s-simpson, 2020-04-19:
https://www.atarimagazines.com/creative/v9n6/114_An_alternative_to_the_Ata.php

The version I had may have been called 2.0P rather than 2.0D, they look sooo similar.  I bet the 2.0P version works correctly

kbr-net, 2020-05-09:
Think, we can close this now.

closed 2020-05-09 by kbr-net