kbr-net/sdrive-max #48: Suggestions for the SDrive.

Carmelojp, 2021-05-01:
"Excuse me, I don't speak English well."

Hello, I was reviewing, and I realized that the CAS file does not emit sound, as it can be configured in ALTIRRA, it would be good to have that option.

It seems to me that it would also be good to be able to read WAV files, although it can be converted from WAV to CAS, it would not hurt to make it complete, or at least to have the NEW option for tape.

I am having trouble loading one type of tape. In Chile they had the famous TURBO SOFTWARE tapes, which an Atari reads normally with any Cassette, in ALTIRRA, configuring they can be loaded, but in the Sdrive-Max, it does not load, "BAD" is read on the screen. 
Link: https://retrogames.cl/foro/viewtopic.php?p=136256

And finally, speaking of Screen, it seems to me that you should take a look at the ILI9481 - ILI9488, They are not selling or arriving 2.8 "screens, only 3.5" screens, not only in my country, if not in many. When installed it looks horrible, seeing it only on a piece of the screen. At least leave the Post open, to see if someone is encouraged to correct the resolution for these screens.

iCityWork, 2021-05-01:
It makes a sound on the computer, just like the original tape drives did,
when loading.

*/Gavin/*

*Gavin Haubelt*

Founder/Lead Developer

iCityWork

(559) 492-6318

***@***.*** ***@***.***>

www.iCityWork.com <http://www.icitywork.com/>

@iCityWork <https://twitter.com/iCityWork>

FaceBook /icitywork <https://www.facebook.com/icitywork>

ᐧ

kbr-net, 2021-09-17:
> "Excuse me, I don't speak English well."
>
> Hello, I was reviewing, and I realized that the CAS file does not emit sound, as it can be configured in ALTIRRA, it would be good to have that option.
>
> It [...]

This is not possible on such a small device with Arduino-UNO. Please do not compare it with Altirra, which has a much more powerfull micro controller system.

> I am having trouble loading one type of tape. In Chile they had the famous TURBO SOFTWARE tapes, which an Atari reads normally with any Cassette, in ALTIRRA, configuring they can be loaded, but in [...]

I will look at it by time.

> And finally, speaking of Screen, it seems to me that you should take a look at the ILI9481 - ILI9488, They are not selling or arriving 2.8 "screens, only 3.5" screens, not only in my country, if not [...]

I do not support other display resolutions, grab the sources and do it by your self(or find one who does it).

kbr-net, 2021-09-19:
> I will look at it by time.

Which file concrete?
I got non of this cas files working neither in emulator, nor in real Atari 600XL/64K/PAL.

kbr-net, 2022-11-04:
Closed due to lake of more information.

closed 2022-11-04 by kbr-net