kbr-net/sdrive-max #51: WAVGAT UNO R3

mamejay, 2021-07-02:
This is a silly question.  I accidentally purchased WAVGAT UNO R3 not realizing that they are not a direct clone.
I have tried to write the eeprom_writer.hex to the UNO and now its stuck in a loop failing on verify.
I cannot reflash any other firmware to the UNO as I get time out issues opening the port.
Any idea if I can make these usable with a SDRive Max and if possible how to recover the boot loop I have on it.

Thanks

StmLord, 2021-08-11:
> This is a silly question. I accidentally purchased WAVGAT UNO R3 not realizing that they are not a direct clone.
> I have tried to write the eeprom_writer.hex to the UNO and now its stuck in a loop [...]

I have about 20 of these UNOs and up till now couldn't find a way to flash them.
Better get a proper atmega 328p clone

mamejay, 2021-08-11:
I ended up desoldering the clone 328 chips and soldering in real ones.  Reflashed using another UNO that worked and now they all work corectly.

kbr-net, 2021-09-17:
You have could used a ISP flash adapter instead.

closed 2021-09-17 by kbr-net