kbr-net/sdrive-max #52: NAK while writing to ATR images

Gitbizy, 2022-01-19:
I purchased an Sdrive Max in December and I am getting NAK errors while trying to copy data from my ATR8000 to this device.  Tried all sorts of things, from changing SIO speed in Spartados 4.49, using MyDos 4.55, etc.  Nothing seems to get rid of the issue, and I can't even copy a full disk (my disks are 3.5" 2880 sector disks on the ATR8000)...  If I chop it up into one file - or sometimes one sub directory  a time, it will copy, but if I try to do a whole disk I will eventually get a NAK error while the SDRIVE is writing. It only happens when writing to the sdrive, never when reading from the ATR.

kbr-net, 2022-02-12:
Can you describe me, how to reproduce this without an ATR8000, so that i can analyze it by my self?

kbr-net, 2025-02-19:
no more response

closed 2025-02-19 by kbr-net