kbr-net/sdrive-max #53: Most newer SD cards does not initialize

StmLord, 2022-05-17:
I've found out the hard way that most newer micro SD cards reports as SD card init failed on the Sdrive, even from the same manufacturer/model/size that few months ago were working great.
up to now Patriot 32GB, kingston 32GB does not work, before I was using the same Kingston canvas select plus 32GB and was working correctly

kbr-net, 2022-05-30:
Most newer, big SD cards are delivered with exFAT formatted by default now. This does SDrive not support. Have you tried to reformat it with FAT32?

TzOk83, 2022-06-05:
Maybe it has something to do with the initialization speed. I've encountered similar problems, when I was trying to use Arduino without a bootloader, and uploaded the firmware directly to MCU, via the AVRdude programmer. Later I found, that the bootloader adds an extra delay after power up, which is required for the SD card, to become initialized and responsive.

kbr-net, 2022-06-08:
Maybe there is also initialisation code missing which is done in bootloader, but an arduino without a bootloader is not an arduino, and i don't want to support such modified variants.

StmLord, 2022-10-21:
It seems to be like timing issues. I've found the time and got 2 different batches of Arduino clones from different sellers and makers and 2 batches of different SD cards. 
1st type of card is Intenso microSDHC UHS-I 16GB ALL (10) of these cards works on every batch and displays I've tested them
2nd type of cards was Patriot microSD 32GB HC-I U1 these cards (10) was hit and miss, in some Arduinos were working as it should and in some others was not readable with "error init SD: 1". Thats the same I had with all newer Kingston 32GB canvas select I had too (5 pieces)
The 2 batches of Arduinos were the same layout with plcc atmega328 with only obvious difference on the location of the RX TX 5v and GND pins. All other ICs, regs etc were the same  (CH340G)
It could be defective UNOs but because the percentage is almost 50-60% on both batches and ALL are working correctly with intenso SD cards it make me think it might be some timing issues on the initialization. 
If the initialisation code is not part of the Sdrive-max firmware then I think its some small difference on caps and resistor tolerances together with very strict timing on the init code (if that's on the UNO boot loader).

kbr-net, 2022-11-04:
This sounds very strange...
Have you compared the FUSE bits on this devices? They affect the startup time also, and it would be interesting, if there are some differences.

TrevorDunne, 2022-11-13:
I had a old 16Gb SanDisk 16GB (2) Micro SD HC I but this into  the SDMAX V1.2 and would not initialize  Fat32
I just thought is the SD card was to slow. so I brought another one Transcend Premium 16GB Micro (1) HCI Read 90MB
and to my shock that did not work as well.
I then tried a another SD Card Transcend Premium 64GB Micro SD (1) XCI Read 90MB and created a partition 16GB Fat32 
using windows inserted SD card into SDMAX v1.2 and now it work fine.
So they must be a problem somewhere is reading SD Cards

kbr-net, 2022-11-14:
Error codes please!

TrevorDunne, 2022-11-14:
On screen
Ready
SIO-Error: 4
SIO-Error: 3

TrevorDunne, 2022-11-14:
I should a put 
Ready
Error init SD: 2
0000007f 00000000

TrevorDunne, 2022-11-14:
This is the Transcend Premium 16GB Micro (1) HCI Read 90MB format as FAT32
[image]

kbr-net, 2022-11-22:
Always the same error on each card?
Is your device powered over the atari, or external?

I will have a look at the initialization code by time, but i need as much as possible information...

Mariuszczs, 2023-12-13:
I have the same problem. Two different boards and two different displays don't work for me.  Error init SD: 1.

Mariuszczs, 2023-12-22:
For a few cards I have 
00000000 0000000
For others
000000ff 00000000

marekl123, 2024-12-14:
I'm experiencing the same problem.

kbr-net, 2025-02-19:
If it is a startup timing problem, one can try to tap and hold on the touchscreen while power up, and do the calibration again. If after this the SD card works...