kbr-net/sdrive-max #56: SDrive_max Screen Display Mirrored ILI9341

overanged, 2022-10-19:
Hi,

Building a SDrive-Max with the UNO R3 Mega328P with a 2.4 Inch TFT ILI9341 screen, I have an issue with the screens display.

Using Xloader v1.00, flashed the boards with V12 and the screen display is mirrored/flipped.

All touch functions work.

I have tried previous versions and V12C3 as availble on the github page.

If anyone has any suggestions, I would be grateful.

Thanks

[image]
[image]
[image]
[image]

vitecd, 2022-10-19:
Did you use this display before? Try Arduino example and check the results,
maybe bad solder joints.

overanged, 2022-10-20:
First time using this display.

New to the Arduino scene, so any help pointing to an easy to load test file would be great.

Joints all look good.

vitecd, 2022-10-20:
try this
https://www.youtube.com/watch?v=EFAfcsYOriM

and try examples
I don't have same library but follow your logic ))
[image: image.png]

overanged, 2022-10-30:
I tried a basic display program and that one was also reversed, so I returned this item back to the seller, got a replacement.

The display was the correct, however after a while the display would go bright, only way to clear was to repower, another faulty display....

This time I am going to another seller, let hope this one works.....

kbr-net, 2022-11-04:
Have you tried the other firmware variants instead of ILI9341?
And have also a look at the TFT-ID, which is displayed on startup very shortly.

kbr-net, 2025-02-19:
No more response, so i will close this.