kbr-net/sdrive-max #64: Possible bug with V 1.4 and older 1.3 that might be related

jfceklosky, 2025-02-07:
I am not able to see all of the files on the root of a FAT 32 formatted SD card with 142 files on it.

I noticed this sort of strange behavior with 1.3.  If I went into a folder and came back out, I could not see all of the file on the root again.

Both of these happen when using the touch screen UI on the sdrive max.  When I boot using DO to see the 'gray' selector, I am able to see all of the files.

There seems to be a bug in the touch screen UI

jfceklosky, 2025-02-07:
Some more details with that same SD card with 140 files on the root.

If I boot using D0 (which loads sdrive.atr (I am running the new one from 1.4)) from the SD card to get to the gray ATR picker and go to the touch screen to see the contents of the SD card. I am able to see all of the file on the root directory on the SD card (showing 1 of 14 pages)

jfceklosky, 2025-02-07:
When using the sdrive.atr from 1.4 to boot from D0 and using the gray screen to look at the sd card, everything is a-ok.

Seems like the touchscreen and sdcard with many file is acting strangely

jfceklosky, 2025-02-08:
**I retested after moving things around on the sd card and it appears to be working now.** 

I don't have so many files on the root directory now.

Firmware 1.4 does fix the directory bug I saw on 1.3 when going in and out of directories.   This was only seem on the TFT screen, not the gray sdrive control program

kbr-net, 2025-02-19:
This was a bug in <1.4 by using menu and display together.
Many files in one directory should be no problem, but it will slow down the menu, especially when going back. I suggest no more than 500 files per directory.

closed 2025-02-19 by kbr-net